roleoroleo/yi-hack-Allwinner #391: Camera config/General/Motion Detection Switch  always show as disabled

ibratos, 2023-02-22:
Firmware Version 0.3.3
Base Version 12.2.0.2_202212191129
Model Suffix y20ga
Serial Number 00009FUS

Camera config/General/Motion Detection  switch  always show as disabled when refresh page, but motion detection works when enabled it and save.

ibratos, 2023-02-22:
http://[IP]/cgi-bin/get_configs.sh?conf=camera
...
"MOTION_DETECTION":"yes",
...

http://[IP]:8080/index.html?page=camera_settings
GUI show Motion Detection disabled

When change anything in camera settings at gui (enable IR for example) it disable Motion detection.

roleoroleo, 2023-02-24:
Fixed:
https://github.com/roleoroleo/yi-hack-Allwinner/commit/8b3ac406ef4fb246b8d787f92bc24a0d063d377d

closed 2023-02-27 by ibratos