roleoroleo/yi-hack-Allwinner #392: Restarts every few minutes since 0.3.3

Madaraxa, 2023-02-22:
Hi. Since i Upgraded to 0.3.3 the camera restarts every Minute. I cant access rtsp stream then and also Not the Web interface. After a few seconds the camera clicks and is back online. Im Not sure if this only happens in night Mode. But at the Moment it is in night Mode.

Madaraxa, 2023-02-22:
As stated dis- and reenabling the rtsp stream seems to worked as a fix

Madaraxa, 2023-02-23:
My second cam (y25) has the same issue. It seems it was random, that it was online for 10 hours. I tried to reflash both cams "offline" now, but it wont succeed as the cams restart too often.
Both cams arent in Nightmode now. so that isnt an issue

ibratos, 2023-02-23:
Has you change it to offline at Yi app ?
The cam stayed like this after going offline in the app and I had to recover a previous Yi image from the SD

Madaraxa, 2023-02-23:
no, I use(d) those two cams for a long time (at least 2 years) with yi-hack. I updated every single update (beside 0.3.0 where i got an error and had to update through ssh) with the OTA update. everything worked fine until 0.3.3...
A (temporary?) fix:
I could reset one of the two cams easily with the reset button on the back. I paired it (like it would be new) with the yicam app. I deleted the cam afterwards in the app and "flashed" the 0.3.3 yi-hack.. Its online for around 6 hours now.
I could NOT pair the second cam with the app. Although the reset worked and the cam connected to WiFi (it said so) the pairing to the yicam app failed (timeout). Then I flashed (at least i think i did) the original yicam firmware. Afterwards I could pair the cam with the app. There it was shown as "offline sinc 2022". i deleted it and repaired it with another name. This worked. Then i deleted the cam again in app and flashed the 0.3.3 completely new. It seemed to work, but when i switched some options in the web interface and saved them, the cam crashed and rebooted... I will keep you up to date