roleoroleo/yi-hack-Allwinner #393: 9fus y31 won't connect to app and webserver isn't running

mfizz1, 2023-02-23:
I was previously on 0.2.6 and decided to update today via webui. Something must have gone wrong because after restart, with the

mfizz1, 2023-02-24:
I have managed to restore an old 8.2.0 firmware and the camera is back up and running. Will update if the hack is succesfull again.

roleoroleo, 2023-02-25:
The home partition was probably full and the cam did not complete the boot procedure.
Restoring the original partition you should be able to hack it again successfully.

mfizz1, 2023-02-25:
Probably, it's working now applied the 0.3.3 hack. But I cannot update to the latest V12 through the app (gets to blinking blue light and that's it nothing happens after that, but that's not a hack issue.

roleoroleo, 2023-02-25:
Same situation probably.
If you want to update, try the unbrick procedure, update and apply again the hack after the update.

mfizz1, 2023-02-25:
I will close this now considering that it is now fixed and not worth exploring as the reasons why this may have happened

closed 2023-02-25 by mfizz1