roleoroleo/yi-hack-Allwinner #394: Snapshot Feature doesnt work

Madaraxa, 2023-02-25:
Hi,
i just tried to enable Snapshot Feature on my cams. But it wont work. I can enable it on the settings page (saved and rebooted cam and its still activated), but the snapshot page of the web interface just keeps "loading" without showing any snapshot.
My other two cameras (with yihack MStar) show a snapshot instant. I reflashed all 4 cameras with stock firmware yesterday, paired them with app, deleted them afterwards and reflashed them with the yiHacks. 
So there shouldnt be a problem :) 
Any hints how to enable snapshot :)?

roleoroleo, 2023-02-25:
What happens if you run the web service?
`http://192.168.0.243:8080/cgi-bin/snapshot.sh`

Madaraxa, 2023-02-25:
It seems it has Problems loading the Page. There is no error message. Its just a blank Page and IT loads and loads and loads....

Madaraxa, 2023-02-25:
Yeah
I tested it. The cam Was online for around 1 hour. I opened the snapshot Page and it crashed and rebootet.

roleoroleo, 2023-02-26:
Is swap file enabled?

Madaraxa, 2023-02-26:
No. I let All settings 'Stock', beside disabling cloud, setting a User and password, renaming the camera and disabling snapshot for recorded Video

roleoroleo, 2023-02-26:
Check the faq page.
The snapshot requires a lot of memory.
Enabling swap file the cam will be stable.

Madaraxa, 2023-02-26:
I will try it. Is there any reason to NOT use swap file? 'you cant' buy sd Cards smaller than 16 GB nowadays. So there should be enough space at any cam :)

roleoroleo, 2023-02-26:
Swap file is not safe for a sd card, after a couple of years the card probably dies.
Not all people want this.

Madaraxa, 2023-02-26:
Hey. With swap file it works. 
Thanks a lot for the help. 
You could add a comment near the snapshot Option in Web interface that swap file is 'needed' for it to work flawless :)

closed 2023-02-26 by Madaraxa