roleoroleo/yi-hack-Allwinner #396: AI Detection not working Y20GA ver: 12.2.0.2_202212191129 0.3.3

Raeprice, 2023-03-18:
I have 2 Yi 1080P Home cameras (Y20GA - BFUS) both have the same config with different host names. Having an issue on only 1 of them where AI Detection is not working.
The only difference between these cameras is the Base version for not working device is 12.2.0.2_202212191129 and the working camera is on 8.2.0.0A_202012041528.
Unsure how to rollback firmware for v12 to 8 to test if this works. Can anyone advise or having similar issues?

roleoroleo, 2023-03-18:
AI detection is completely new in the last fw (12).
Check if you configure the cam properly.
I'm currently use a y20ga with fw 12 and it's ok.

Raeprice, 2023-03-20:
> AI detection is completely new in the last fw (12). Check if you configure the cam properly. I'm currently use a y20ga with fw 12 and it's ok.

So i havent been able to get this going. I copied the config from the working device on V8 onto the V12 device and changed hostname and for some reason the human detection still not reporting. Im using Motion detection which seems to work well. It will do for my purpose for now. Will keep looking at this and report back if i find a solution

closed 2024-05-05 by github-actions[bot]