roleoroleo/yi-hack-Allwinner #400: yi home and motioneye

Bodengriller, 2023-05-25:
I'm desperately trying to integrate my three yi-home cameras into motioneye in order to realize a usable motion detection.
The whole thing fails already because the rtsp (high) stream does not arrive reasonably in motioneye.
Only fragments are displayed - no matter which setting I try.
[image]

Is the camera simply too weakly equipped for this?
I have already deactivated all services that are not needed (including snapshot):
[IP]/cgi-bin/get_configs.sh?conf=camera:

```
"SWITCH_ON":"yes",
"SAVE_VIDEO_ON_MOTION":"yes",
"SENSITIVITY":"low",
"AI_HUMAN_DETECTION":"no",
"SOUND_DETECTION":"no",
"SOUND_SENSITIVITY":"80",
"LED":"no",
"ROTATE":"no",
"IR":"no",
"CRUISE":"no",
"MOTION_DETECTION":"no",
"AI_VEHICLE_DETECTION":"no",
"AI_ANIMAL_DETECTION":"no",
"HOMEVER":"8.2.0.0A_202012041528",
"NULL":"NULL"
```

[image]
[image]
[image]

I've often read here that the snapshot function probably eats a lot of resources and should be disabled if possible if you need more performance.
Why is that?
As I understand it, the camera only takes a screenshot and makes it available - doesn't sound very complex.

When I open the RTSP stream via VLC on my computer, it strangely works fine.
This made me suspect that it was related to motioneye itself.
That's why I installed a new motioneye installation on my computer (as Ubuntu VM) -> same problem as with the motioneye installation on my Futro S740 homeserver in the proxmox container.

Same Problem with these three yi-Home-Models:
1. IFUS Y654GT	9.0.36.00_202011101611	y211ga	yi-hack-Allwinner-v2	0.2.6
2. 9FUS Y31SAO	8.2.0.0A_202012041528	y20ga	yi-hack-Allwinner		0.3.3
3. 9FUS Y31RW6	8.2.0.0A_202012041528	y20ga	yi-hack-Allwinner		0.3.3

With which parameters (FPS, resolution, ...) does the RTSP-High-stream stream? (1920x1080; ? FPS)

Right off the bat, it doesn't look like the network is overloaded from the 3 cameras...I couldn't imagine that either.
Nevertheless, does anyone know what traffic the RTSP-high-stream causes or needs?

any idea what else I could test?

Thanx

roleoroleo, 2023-06-03:
Sorry for the delay.
I see you solved it but I'll answer anyway.

> Is the camera simply too weakly equipped for this?

The cam should be able to stream hd video without problems.

> I've often read here that the snapshot function probably eats a lot of resources and should be disabled if possible if you need > more performance.
> Why is that?
> As I understand it, the camera only takes a screenshot and makes it available - doesn't sound very complex.

This is true.
The reason is that this feature is not available in the original fw. I added it in the hack
And since I don't have access to the hw codec, I grab an iframe from the buffer and I convert it to jpeg with ffmpeg libraries.
This operation requires 2-3 MB of memory and the cam runs with very low memory available.

> When I open the RTSP stream via VLC on my computer, it strangely works fine.
> This made me suspect that it was related to motioneye itself.

Is it possibile to configure motioneye with tcp instead of udp?

> With which parameters (FPS, resolution, ...) does the RTSP-High-stream stream? (1920x1080; ? FPS)

rtsp
1920x1080
20 fps
gov 40

> Nevertheless, does anyone know what traffic the RTSP-high-stream causes or needs?

VLC shows about 500 kbit/s

Bodengriller, 2023-06-03:
Thank you for clarifying the unanswered questions.
In the meantime I'm playing around with Frigate.
Motioneye is good but still brings me too many false-positives.

One last question:
What is your experience with RTSP Stream Low + High enabled?
Does it work acceptably over time?

I would like to use the low-res stream for motion detection, but record in HD.

roleoroleo, 2023-06-04:
> What is your experience with RTSP Stream Low + High enabled - does it work acceptably over time?

I don't use the cam with a DVR, so I don't know if the stream is stable over a long period.
But it should work.

> Wouldn't it be nice to make the streamed FPS adjustable in yi-hack to reduce network-traffic? :)

AFAIK it's not possible to change codec parameters.

closed 2024-05-05 by github-actions[bot]