roleoroleo/yi-hack-Allwinner #402: Please add FTPS support and ability to specify port for push

divadiow, 2023-06-18:
No idea how much work this would involve, but could FTP over TLS be added please and the ability to specify the port? This is for uploading to an FTP not the FTP server that can be run on the camera.

Also, what does the existing FTP support? plain text basic only? passive?

roleoroleo, 2023-06-18:
What do you mean.
FTP server inside the cam or FTP client to upload records?

divadiow, 2023-06-18:
FTP client to upload recordings

roleoroleo, 2023-06-19:
I'm trying to build lftp but there are too many dependencies.

divadiow, 2023-06-19:
Bum :(

Is lftp the only option?

roleoroleo, 2023-06-19:
I don't know.
I will try again.

roleoroleo, 2023-06-20:
I successfully compiled lftp, it was a bloodbath. Unfortunately it's a big binary (~4 MB) and I can't add it to the hack.
If you want, I can share. You could install it in your sd card and create a custom script.

divadiow, 2023-06-22:
yes please!

although at the moment I appear to have done something to my cam to make it reboot every 15s or so. yihack remove run several times, re-hack tried, resets etc. it's stable until I pair it with the app, as a virgin camera with no hack, then it starts looping.

so I need to fix that first! :sigh:

roleoroleo, 2023-06-22:
If you already restored the home partition, the problem could be the power supply.
Try to change it.

Here lftp:
[lftp.gz](https://github.com/roleoroleo/yi-hack-Allwinner/files/11830156/lftp.gz)

divadiow, 2023-06-22:
thanks for the ftp.

indeed. I've tried different chargers and cables. all the same :(

roleoroleo, 2023-06-22:
Did you restore the home partition with the unbrick procedure?

divadiow, 2023-06-22:
hmm. doesn't like my home partition from backup on another cam. don't suppose you have original whole firmware for y20ga 9FUS I can factory reflash everything?

divadiow, 2023-06-22:
Ok sure. I guess I just wanted a virgin home partition. To clarify, what actually happened is that the flashed partition from the other y20ga (BFUS - does this matter? As long as it's still from a y20ga?) did work, but it had the yihack from the other cam on it. I tried to dehack (using correct method) but this put it back to how it was, rebooting every 15s. I wanted to be sure I could unhack successfully in the event of future firmware releases from Yi.

I actually then reflashed the cam with the backup but this time it was boot looping again, whereas the first time it seemed to fix it. So, dunno

roleoroleo, 2023-06-22:
It's not a problem if it's a BFUS or other, the important thing is the suffix y20ga.

Probably the problem is the following:
The bootloader, when it restores the partitions, saves the checksum of the file and doesn't load the file anymore if it already loaded.
It's a strategy to avoid a boot loop.
You need a file with at least 1 byte different.
I will update the unbrick procedure fot this purpose.

closed 2024-05-05 by github-actions[bot]