roleoroleo/yi-hack-Allwinner #403: Hardware donation

roleoroleo, 2023-06-23:
I'm having trouble maintaining support for all cams because I have very few models available. Especially older ones that are no longer available for purchase.
If you have cameras that you don't use anymore (maybe broken) and you want to get rid of them, please send to me.

essys, 2023-10-27:
Can you provide a list of cams you need to continue to support such models?
Or vice versa, a list of the cams that you have.
The destination country would be Spain?

roleoroleo, 2023-10-27:
About allwinner I have just a y20ga.

roleoroleo, 2024-04-03:
Good idea.
I will do it.

essys, 2024-04-12:
Please do so, even though I already donated a few euros last year, I would like to support you in the further development of this great mod and ideally extend the device support.

closed 2024-08-11 by github-actions[bot]