roleoroleo/yi-hack-Allwinner #404: Topics Prefix for remote configuration not works

snakuzzo, 2023-06-25:
Hi, I've two YI Camera 1080p (y20ga) with latest firmware and latest hack (v.0.3.4).
I'm trying to set "Topics Prefix for remote configuration", but when I set and save my topic, I always see default settings...

[image]

Tried to reboot...same result

roleoroleo, 2023-06-25:
Maybe a js problem.
Try to clean the cache of the browser.

snakuzzo, 2023-06-25:
Cleaned cache...tried using Chrome, Firefox, Edge...same result

roleoroleo, 2023-06-26:
Sorry, I temporarily removed this feature in this commit: https://github.com/roleoroleo/yi-hack-Allwinner-v2/commit/b6fad82867cead2ea10834fa2a9cb5b548b5b4e8
In this moment it's not working.
I will restore it in the next release.

snakuzzo, 2023-06-26:
Ok. Thak you!

snakuzzo, 2023-07-31:
Hi @roleoroleo.
Any news about this feature ?

roleoroleo, 2023-08-18:
Fixed here: https://github.com/roleoroleo/yi-hack-Allwinner-v2/commit/c5ad71bd08de718e41b7e73ceaa04a00eb7499e2

closed 2024-05-05 by github-actions[bot]