roleoroleo/yi-hack-Allwinner #405: [Feature] Sending person detection information to NVR

WMP, 2023-06-25:
Hi, I recently bought a Reolink camera that has functions to detect people, vehicles and animals. After adding it in synology surveillance station using ONVIR, I have the ability to mark when NVR should save the recording. There are options to choose from in which the camera is responsible for detecting a person/vehicle/animal and sends such information to the NVR.

[image]

Yi cameras have the same detection function, but I can't configure the synology surveillance station to start recording when the camera recognizes a person. Could it add such an option?

roleoroleo, 2023-06-26:
It's not simple to add this feature.
I already tried several yeats ago but I failed.
Now there is a new onvif service, so I could try again.
But it's not a short task.

roleoroleo, 2023-06-26:
I completely rewritten a new onvif daemon.
It's a very simple and stupid server that detect the request and send the response based on a xml template.
The damon is in plain C (the old one was in C++ with XML/GSOAP engine).

The core of this patch is to add event subscription support based on this specification:
https://www.onvif.org/wp-content/uploads/2016/12/ONVIF_WG-APG-Application_Programmers_Guide-1.pdf
Probably using WS-BaseNotification.

WMP, 2023-06-26:
Unfortunately I don't know c, so I cannot write this. But i think it should be simple, if it is what is write on 
> 9.3 Setting Up WS-BaseNotification 

If I understand this right, camera should only receive address from NVR to send on this address post about detect people. I try to make simple bash script to send this post.

roleoroleo, 2023-06-26:
Onvif works as a cgi with httpd.
So you can write your own script and add to the folder.

WMP, 2023-06-26:
Hmmm ;) So only, please told me how to trigger my script when camera detect people.

WMP, 2023-06-26:
I tcpdump my Reolink camera and this works in other way. My NVR ask even 1 second my camera about alarms:
```
===================================================================
Follow: tcp,ascii
Filter: tcp.stream eq 65
Node 0: NVR:60824
Node 1: YI:80
166
GET /cgi-bin/api.cgi?cmd=GetAiState&token=b1b4a92298310ba&channel=0 HTTP/1.0
User-Agent: Synology Surveillance Station
Authorization: Basic XXXXXX

        668
HTTP/1.1 200 OK
Date: Mon, 26 Jun 2023 20:03:37 GMT
Content-Type: text/html
Connection: close
X-Frame-Options: SAMEORIGIN
X-XSS-Protection: 1; mode=block
X-Content-Type-Options: nosniff

[
   {
      "cmd" : "GetAiState",
      "code" : 0,
      "value" : {
         "channel" : 0,
         "dog_cat" : {
            "alarm_state" : 0,
            "support" : 1
         },
         "face" : {
            "alarm_state" : 0,
            "support" : 0
         },
         "people" : {
            "alarm_state" : 1,
            "support" : 1
         },
         "vehicle" : {
            "alarm_state" : 0,
            "support" : 1
         }
      }
   }
]

===================================================================

```

In this HTTP GET from NVR to YI:80/cgi-bin/api.cgi?cmd=GetAiState&token=b1b4a92298310ba&channel=0 , camera response about detect people: 
         "people" : {
            **"alarm_state" : 1,**
            "support" : 1
         },

So we only need to support **token=b1b4a92298310ba** param and make script to write file with `"alarm_state" : 1,` when camera detect people, or  `"alarm_state" : 0,` when camera dont detect people.

WMP, 2023-06-26:
From what I have read about onvif, we should use profile m: https://www.onvif.org/profiles/profile-m/ because profile S does not support video analysis: https://www.onvif.org/profiles/profile-s/ . Unfortunately, synology does not support profile M, only profile S. On a quick search I found that Dahua NVR seems to support profile M. 

However, if you wanted to implement an M-profile, you don't necessarily have to use WS-BaseNotification, you could use Pull-Point, and it would work similar to the Reolink protocol above. It seems to me that Pull-Point would be easier than configuring WebServices(?).

roleoroleo, 2023-06-27:
I think you can use Profile-S as well.
And yes, probably Pullpoint and WS-BaseNotification are equivalent for this purpose.
Push vs pull...

To detect an event, you can:
1 - Read the message queue, see here: https://github.com/roleoroleo/mqttv4/blob/master/src/mqttv4.c
or
2 - Subscribe to mqtt topic where human detection are published.

closed 2024-05-05 by github-actions[bot]