roleoroleo/yi-hack-Allwinner #407: 0.3.5 Failed setup for HA

AJStubbsy, 2023-07-15:
Hi 
My Camera is a Yi Home 1080p
BFUS type
Using the y20ga firmware

I tried HA ONVIF integration and had the following 'Warning': -

```Logger: homeassistant.config_entries
Source: config_entries.py:432
First occurred: 19:11:07 (1 occurrences)
Last logged: 19:11:07

Config entry 'YiCamTest - 38:01:46:d0:2d:9a' for onvif integration not ready yet: Could not connect to camera: ActionNotSupported: <Element {http://www.w3.org/2003/05/soap-envelope}Detail at 0x5d01d868> (code:env:Receiver) (subcodes:{http://www.onvif.org/ver10/error}ActionNotSupported); Retrying in background
```

[image]

Maybe this log is of some use to you?

By the way, I have the camera integrated as a Generic Camera and was having some connection issues where the snapshot kept going black and then coming back again. Since upgrading to the 0.3.5 firmware, it's been a far more stable connection to the camera. I can't really see a benefit of me going back to the ONVIF integration but I thought I'd let you know anyway.

Thanks

roleoroleo, 2023-07-15:
Apart from the warning, does the onvif integration work?
Did you try my integration?
https://github.com/roleoroleo/yi-hack_ha_integration

And another problem: 0.3.5 is not yet released, probably you managed to download it while i was preparing the release page.
This shouldn't be a problem, but you should reinstall it when I will release.

AJStubbsy, 2023-07-16:
No it doesn't work. I thought it was strange that it shows as a warning rather than an error because it doesn't actually create the entity.

I installed the pre-release of 0.3.5; something that was so easy I was shocked! I thought I'd have to faff about with reconnecting the camera but obviously your firmware doesn't overwrite the wifi settings, etc. Superb!

TBH I've got the camera working with Generic Camera and it's fine for my purposes, although it's great to know your integration exists, if I have problems in the future.

As always, thank you so much for this fantastic firmware and for your readiness to respond and find solutions to problems, it's very much appreciated.

AJStubbsy, 2023-07-17:
I can confirm that everything is now working again with 0.3.6.

Many thanks

closed 2023-07-17 by AJStubbsy