roleoroleo/yi-hack-Allwinner #408: Camera unresponsive after firmware update from 0.3.4 to 0.3.6

othyro, 2023-07-20:
After updating the camera firmware from 0.3.4 to 0.3.6, it's unresponsive over the network. Using the [configure wifi](https://github.com/roleoroleo/yi-hack-Allwinner-v2/wiki/Change-WiFi-credentials-(deprecated)) script doesn't seem to work either. The orange LED remains lit.

If I remove the SD card and hold the reset button, it goes through reset mode and I hear, 'Waiting to connect...'. If I reboot it, I can hear the shutter clicking twice before it awaits a connexion. Reinserting the SD card before replugging in the USB cable, leaves the orange LED lit, I hear no shutter, and the camera doesn't connect to the WiFi.

Not sure how to proceed. I'm not using the app nor want to do so. Here's what the status page looked like after I did the initial flash some weeks ago:
[image]
I want to upgrade the firmware on the other cameras too, since I have two more of this model. However, I don't want the same to happen again. What logs would you like to see?

roleoroleo, 2023-07-21:
Please, follow the unbrick procedure.
https://github.com/roleoroleo/yi-hack-Allwinner-v2/tree/master/unbrick
using the "hacked" option.
And apply again the hack.

othyro, 2023-07-21:
Thanks. It seems to have been my fault for not having cleared all the motion events before updating it. The camera's working properly now.

closed 2023-07-21 by othyro

othyro, 2023-10-28:
[image]
On one of my other cameras with this firmware, I'm not seeing the upgrade version although several are available. Is there any way to force the update via the CLI?

roleoroleo, 2023-10-28:
Yes, you can download the file manually and copy it to the root folder of your sd card.
Then, rename it suffix_x.x.x.tgz, for example y20ga_x.x.x.tgz
Finally reboot the cam.

othyro, 2023-10-28:
The _/_ partition is mounted read-only. I tried with the file in _/home/yi-hack_, _/tmp_, _/tmp/sd_, and _/tmp/sd/yi-hack_. The device didn't update in any of those cases. What's the correct path if I'm connected via SSH?

roleoroleo, 2023-11-03:
The correct path is /tmp/sd where the cam should mount your sd card.
If the sd card is mounted read only, remove it and try a file system check with a PC.
Then try again.

othyro, 2023-11-03:
I reuploaded the file to _/tmp/sd_, rebooted, and  it's running 0.3.4 still. Any advice on how to proceed?

roleoroleo, 2023-11-08:
After uploaded the file and rename, you have to click "Upgrade firmware" in the web gui.