roleoroleo/yi-hack-Allwinner #409: Some problems

ethempekin, 2023-07-25:
Today started a big problem with my cameras.
Here related:
Firmware Version | 0.3.6
Base Version | 8.2.0.0A_201912270941

I can access the camera only via the IP address but no snapshot available.  One homebridge the following error happens: 
FFmpeg exited with code: 1 and signal: null (Error)

I have additional difficulties with Yi Home app and this is likely to be due to the latest iOS update.

roleoroleo, 2023-07-26:
Test the cam with vlc and check if the stream works.

ethempekin, 2023-07-26:
Thank you. With some efforts - resetting. I am able to sort the issue for most of my cameras, except for Yi 1080p Dome. That one is covered with another hack.

closed 2024-05-04 by github-actions[bot]