roleoroleo/yi-hack-Allwinner #410: HA Fetch snapshot image failed.

AJStubbsy, 2023-08-02:
Yi Home 1080p
y20ga_0.3.6

Hi, do you have any idea why I keep getting the following ERROR in Home Assistant please?

```Fetch snapshot image failed from YiCam, falling back to FFmpeg```

I have snapshot enabled on the camera itself.
Thanks

AJStubbsy, 2023-08-05:
By the way. I'm using ONVIF integration. I did try your integration but no matter what I did the camera was shown as OFF. I noticed some others were experiencing the same problem, so I reverted to the ONVIF integration, since the video stream is fine and that's all I really need in HA. It's just the ERROR that's odd.

roleoroleo, 2023-08-19:
To debug the problem, try to call a snapshot with the webui.
http://IP_OF_THE_CAM/?page=snapshot
or
http://IP_OF_THE_CAM/cgi-bin/snapshot.sh?res=high

AJStubbsy, 2023-08-20:
Maybe it's just HA wanting the image sooner than it loads because I successfully get a snapshot using either address that you suggested.

Thanks for taking the time to respond, I appreciate it's probably on HA side, rather than on yours but I also know you've got a better idea than me as to what can be the problem ;)

closed 2024-05-04 by github-actions[bot]