roleoroleo/yi-hack-Allwinner #413: Kami Baby Support

osh123, 2023-09-08:
Continuing the [ticket](https://github.com/roleoroleo/yi-hack-Allwinner-v2/issues/705) in the [roleoroleo/yi-hack-Allwinner](https://github.com/roleoroleo/yi-hack-Allwinner) project.

I would like to test a beta version for Kami Baby.
Serial number is starting with LFUS, firmware version is 1.3.0.0C.
Firmware backup: https://drive.google.com/file/d/1rSXdw0ar4VVb0X1-tJpdtiU8fOVGA9sa/view?usp=sharing
Also, this camera has temperature and humidity sensors, any chance I can read them and expose them to Home Assistant integration? 
The camera also has night vision support, will it also work? will it be possible to control it?

Thanks a lot!

roleoroleo, 2023-09-08:
Please, try this new beta firmware:
[y501gc_0.3.6.tar.gz](https://github.com/roleoroleo/yi-hack-Allwinner/files/12560591/y501gc_0.3.6.tar.gz)

> Also, this camera has temperature and humidity sensors, any chance I can read them and expose them to Home Assistant integration?

At the momento it's not possible, and without a cam under my hands it's very difficult to develop this feature.

> The camera also has night vision support, will it also work? will it be possible to control it?

You can control the enable/disable, but when you enable it the cam will switch on/off automatically.

osh123, 2023-09-10:
Hi, 

Thank you for the firmware!
I've managed to load it into the camera, but I'm unable to view the rtsp stream or take a snapshot.

osh123, 2023-09-10:
Adding the I was able to see the temperature and humidity value with `dmesg`:
```
[  133.684747] ----sht3x temperature(26833), humidity(56146)----
root@yi-d203:~/bin#
```

roleoroleo, 2023-09-20:
> I Wrote a small script that fetches the temp data from dmesg and publishes it in MQTT:

Ok, I will try to add to the hack.

> More than that, the camera also has a light that i would really like to control from HA, I've ran `ipc_read` while toggling the light switch in app, the strange thing is that I'm getting the same message for both on and off, does it make sense to you?

Probably the message you sniffed is not related to this function but you can try.

> If so, I'll try to compile `ip_cmd` with it and try to run it

It's not necessary, you can load a message from file with ipc_cmd.
Save the file in binary mode with:
`echo -ne '\x40\x00 ....' > /tmp/sd/message.bin`

> can it cause damage to the camera?

I can't be sure but it's never happened to me.
And I have executed many commands...

osh123, 2023-09-22:
> About the stream, please send me a copy of your buffer /dev/shm/fshare_frame_buf

https://drive.google.com/file/d/1-K7e7LzqW91J6t0hKw_Dx4UXc3wZGa-z/view?usp=sharing

> It's not necessary, you can load a message from file with ipc_cmd.
> Save the file in binary mode with:
> `echo -ne '\x40\x00 ....' > /tmp/sd/message.bin`
That's great! Thanks!
I didn't work with that message, but now it will be easier to research.

osh123, 2023-09-22:
Another update: 
The correct message for light on is:
`\x01\x00\x00\x00\x08\x00\x00\x00\x22\x40\x01\x00\x01\x00\x00\x00\x01`
For light off its:
`\x01\x00\x00\x00\x08\x00\x00\x00\x22\x40\x01\x00\x01\x00\x00\x00\x00`

The light also has an option to control the brightness and it has a physical button as well.  
I'll try to compile the code and implement it as a light entity in HA.

Thanks for all the help.

roleoroleo, 2023-09-23:
> Another update: The correct message for light on is: `\x01\x00\x00\x00\x08\x00\x00\x00\x22\x40\x01\x00\x01\x00\x00\x00\x01` For light off its: [...]

Great! Keep me updated.
About the stream, try to overwrite the standard binaries with the content of this archive.
It should work, both rtsp and snapshot.
[kami_stream_prg.tar.gz](https://github.com/roleoroleo/yi-hack-Allwinner/files/12706135/kami_stream_prg.tar.gz)

osh123, 2023-09-23:
> Great! Keep me updated.
> About the stream, try to overwrite the standard binaries with the content of this archive.
> It should work, both rtsp and snapshot.
> [kami_stream_prg.tar.gz](https://github.com/roleoroleo/yi-hack-Allwinner/files/12706135/kami_stream_prg.tar.gz)

Still, doesn't stream either RTSP or snapshot.
Running the RTSP (and h264grabber_h) server with `-d 3` gives the following error:
```bash
root@kami-baby:~/bin# rRTSPServer -m y501gc -r high -d 3
1695465909804: the size of the buffer is 1786224
1695465909805: mapping file fshare_frame_buf, size 1786224, to b6c11000
1695465909806: closing the file fshare_frame_buf
1695465909806: starting capture main loop
1695466018218: warning - -1 frame(s) lost
```

roleoroleo, 2023-09-23:
h264grabber is not needed anymore to run rRTSPServer.

It seems that rRTSPServer is not able to detect the type of stream.
Please try low res stream.

osh123, 2023-09-23:
> h264grabber is not needed anymore to run rRTSPServer.
> 
> It seems that rRTSPServer is not able to detect the type of stream. Please try low res stream.

Same result with the low res stream.
It doesn't seem to even listen to 554.
Any other logs that could be useful for you?

osh123, 2023-09-23:
https://drive.google.com/file/d/1BDX-IPlz9tgViZYksVgZZOIKqz1OoO2E/view?usp=sharing
This is another stream i captured (last one was with `nc`, maybe it was incorrect?)
Any other way you recommend to capture it?

Thanks!

roleoroleo, 2023-09-23:
Ok, try this one:
[kami_stream_prg_2.tar.gz](https://github.com/roleoroleo/yi-hack-Allwinner/files/12706830/kami_stream_prg_2.tar.gz)

osh123, 2023-09-24:
> Ok, try this one: [kami_stream_prg_2.tar.gz](https://github.com/roleoroleo/yi-hack-Allwinner/files/12706830/kami_stream_prg_2.tar.gz)

Seems to be working!
low res is working smoothly (both stream and snapshot).
high res stream is showing a gray screen every 10 secs or so, and snapshot is not working.
But that's good for me :)
Could it be that the high res stream is heavy on the camera resources?

Thank you!

roleoroleo, 2023-09-24:
> Could it be that the high res stream is heavy on the camera resources?

Maybe but I will check again. High res snapshot should work.
Did you enable swap space?

roleoroleo, 2023-09-24:
Oh, I found a problem in the ffmpeg compilation flags.
Try this:
[imggrabber.gz](https://github.com/roleoroleo/yi-hack-Allwinner/files/12708880/imggrabber.gz)

osh123, 2023-09-24:
> Oh, I found a problem in the ffmpeg compilation flags. Try this: [imggrabber.gz](https://github.com/roleoroleo/yi-hack-Allwinner/files/12708880/imggrabber.gz)

Worked :)
Thank you!

roleoroleo, 2023-09-25:
I will add Kami Baby to the next release.

osh123, 2023-09-25:
Can you wait for my light control implementation? 🙃

roleoroleo, 2023-09-25:
Yes, no problem.
Could you check if temperature and humidity values are available in any ipc message?

osh123, 2023-09-25:
Will do 😃

osh123, 2023-09-26:
Another small issue with the snapshot service: calling snapshot.sh with `watermark=no` causes the snapshot to fail.

roleoroleo, 2023-09-26:
This is strange.
Did you try 2 or 3 times?
It may happen that the cam is busy and returns error.

osh123, 2023-09-26:
Tried multiple times, didn't work.
Also, the high res stream seems very unstable, I'm getting a gray screen every few seconds.
VLC is reporting about a lot of lost frames: around 57 frame for 20 seconds of running.

osh123, 2023-09-26:
Sure, I can test it.
It will be compatible with my model?

osh123, 2023-10-01:
Hi,

Installed the new version, still seeing some hangs.
Any logs i can supply to help you understand the problem? 

Thanks!

roleoroleo, 2023-10-01:
Yes, try to run this version with -d 15.
Connect the client and post the output.

osh123, 2023-10-01:
See here: 
https://drive.google.com/file/d/1gX95aNLal256TpZQGAPFbYJ8GLn5XXRW/view?usp=sharing

roleoroleo, 2023-10-01:
Apart from a few lost frames at the beginning there are no errors in the log.
All frames are properly processed.
Please try:
- using ffmpeg as a client - `ffmpeg -i rtsp://IP/ch0_0.h264 test.mp4`
- running the daemon without audio.

roleoroleo, 2023-10-09:
I will check the code.

Is low res stream ok?

osh123, 2023-10-17:
Low res stream seems stable.

roleoroleo, 2023-10-19:
If you can, please try a test with the camera close to the router.
Did you enable swap space?

roleoroleo, 2023-10-25:
Please, try this binary:
[rRTSPServer.gz](https://github.com/roleoroleo/yi-hack-Allwinner/files/13167358/rRTSPServer.gz)

osh123, 2023-10-26:
> If you can, please try a test with the camera close to the router. Did you enable swap space?

I tried testing it close to the router - still it happens - gray screen.
swap space is enabled.

> Please, try this binary: [rRTSPServer.gz](https://github.com/roleoroleo/yi-hack-Allwinner/files/13167358/rRTSPServer.gz)

It seems much more stable, the gray screens are much less frequent.

Thank you

roleoroleo, 2023-10-27:
Ok, send me a log of the rtsp server.
Stop the watchdog, stop the server and start the server in debug mode:
```
killall wd_rtsp.sh
killall rRTSPServer
rRTSPServer -m y501gc -r high -a no -p 554 -d 3
```
Finally, post the output.

osh123, 2023-10-31:
[https://drive.google.com/file/d/1Y5CXaQ8jzB_5Ky4cg2dN_XP4Dh5ihNTM/view?usp=sharing](kami-baby-rtsp-log.txt)

closed 2024-05-04 by github-actions[bot]

weilinear, 2024-07-28:
Reopening this, thanks for having the beta, somehow I used the latest release for the baby camera but I don't seem to see the Low Res streams or the Audio Only stream. Only the high res stream and has the same gray screen every ~10s issue.

roleoroleo, 2024-07-30:
Please, send me a copy of your frame buffer file fshare_frame_buf