roleoroleo/yi-hack-Allwinner #414: Unbrick Cam

neddyuk, 2023-09-08:
I am looking for the solder points to solder a usb connection onto the pads of the Yi Dome X cam. I saw someone had sucess on [this page](https://github.com/roleoroleo/yi-hack-MStar/issues/262) bbut they seem to have changed the mainboard and chip now so the pads arent labelled . Anyone have any ideas on the location UART pads? images below 

[image]
[image]

roleoroleo, 2023-09-08:
You could try to follow UART pins.
https://www.senslab.com/uploads/20220406/da212226e7ec445d6b7127a27101d262.pdf

neddyuk, 2023-09-08:
> You could try to follow UART pins. https://www.senslab.com/uploads/20220406/da212226e7ec445d6b7127a27101d262.pdf

Not sure which ones those would be to start with. Otherwise yea i could use the continuity mode on the multimeter. Seems the board is from 2019 so has been out a while has it not be stumbled across before?

closed 2024-05-04 by github-actions[bot]