roleoroleo/yi-hack-Allwinner #419: v0.37 audio fail

ibratos, 2023-11-02:
Firmware Version | 0.3.7
Base Version | 12.2.0.2_202212191129
Model Suffix | y20ga
Serial Number | 00009FUS

After upgrade to v.37 audio doesn't work.
Whith kami app it works.
Rtsp: alternative server On
Rtsp stream low
Rtsp audil  aac

From VLC
rtsp://IP/ch0_1.h264   Login Ok. Video Ok, Audio No Ok
rtsp://IP/ch0_2.h264   Login Ok, Not responde.

ibratos, 2023-11-03:
Update info
Without rtsp alternative (live555) works fine video and sound.

roleoroleo, 2023-11-08:
I will check the code.

roleoroleo, 2023-11-09:
What format are you using with the alternative server?
aac? ulaw? ...

ibratos, 2023-11-09:
AAC

roleoroleo, 2023-11-09:
Please try this binary:
[rtsp_server_yi.gz](https://github.com/roleoroleo/yi-hack-Allwinner/files/13308435/rtsp_server_yi.gz)

ibratos, 2023-11-09:
Hi, it doesn't change.
This binary with AAC doesn't give audio, in both urls ( ch0_1.h264 and ch0_2.h264)
regards

roleoroleo, 2023-11-10:
Tested now with the same model (y20ga fw12) and it works.
Which client are you using?
Please try to use ffmpeg and send me the log:
`ffmpeg -i rtsp://IP_CAM/ch0_0.h264 test.mp4`

roleoroleo, 2023-11-10:
Maybe I found.
Try this:
[y20ga_0.3.7.tar.gz](https://github.com/roleoroleo/yi-hack-Allwinner/files/13319727/y20ga_0.3.7.tar.gz)

ibratos, 2023-11-12:
Hi, with this new release it works

closed 2023-11-17 by ibratos

drakodga, 2023-11-17:
Hi. Can you upload new 0.38 version with audio fix? Thank you!

roleoroleo, 2023-11-18:
I'm far from a new release.
Which model do you have?

drakodga, 2023-12-11:
Sorry for the delay. The information you have requested from me is this:
[image]

roleoroleo, 2023-12-11:
You can user the version posted above.

drakodga, 2023-12-11:
I tried it, but with same result. Finally, i'am in previous version that it's fine.