roleoroleo/yi-hack-Allwinner #421: audio doesn't work

xcoplit, 2023-11-24:
Firmware Version: 0.3.7
Base Version: 8.2.0.0A_201912270941
Model Suffix | y20ga
Serial Number | BFCN****

audio doesn't work.
Whith yi-home app it works.

From VLC
rtsp://IP/ch0_0.h264 Video Ok, Audio No Ok

ffmpeg -i rtsp://IP_CAM/ch0_0.h264 test.mp4
---------------------------------------------
ffmpeg version 4.4.2-0ubuntu0.22.04.1 Copyright (c) 2000-2021 the FFmpeg developers
  built with gcc 11 (Ubuntu 11.2.0-19ubuntu1)
  configuration: --prefix=/usr --extra-version=0ubuntu0.22.04.1 --toolchain=hardened --libdir=/usr/lib/x86_64-linux-gnu --incdir=/usr/include/x86_64-linux-gnu --arch=amd64 --enable-gpl --disable-stripping --enable-gnutls --enable-ladspa --enable-libaom --enable-libass --enable-libbluray --enable-libbs2b --enable-libcaca --enable-libcdio --enable-libcodec2 --enable-libdav1d --enable-libflite --enable-libfontconfig --enable-libfreetype --enable-libfribidi --enable-libgme --enable-libgsm --enable-libjack --enable-libmp3lame --enable-libmysofa --enable-libopenjpeg --enable-libopenmpt --enable-libopus --enable-libpulse --enable-librabbitmq --enable-librubberband --enable-libshine --enable-libsnappy --enable-libsoxr --enable-libspeex --enable-libsrt --enable-libssh --enable-libtheora --enable-libtwolame --enable-libvidstab --enable-libvorbis --enable-libvpx --enable-libwebp --enable-libx265 --enable-libxml2 --enable-libxvid --enable-libzimg --enable-libzmq --enable-libzvbi --enable-lv2 --enable-omx --enable-openal --enable-opencl --enable-opengl --enable-sdl2 --enable-pocketsphinx --enable-librsvg --enable-libmfx --enable-libdc1394 --enable-libdrm --enable-libiec61883 --enable-chromaprint --enable-frei0r --enable-libx264 --enable-shared
  libavutil      56. 70.100 / 56. 70.100
  libavcodec     58.134.100 / 58.134.100
  libavformat    58. 76.100 / 58. 76.100
  libavdevice    58. 13.100 / 58. 13.100
  libavfilter     7.110.100 /  7.110.100
  libswscale      5.  9.100 /  5.  9.100
  libswresample   3.  9.100 /  3.  9.100
  libpostproc    55.  9.100 / 55.  9.100
[rtsp @ 0x55c1ee52b680] max delay reached. need to consume packet
[rtsp @ 0x55c1ee52b680] RTP: missed 34 packets
[h264 @ 0x55c1ee530b00] error while decoding MB 84 45, bytestream -15
[h264 @ 0x55c1ee530b00] concealing 2725 DC, 2725 AC, 2725 MV errors in I frame
Input #0, rtsp, from 'rtsp://192.168.32.23/ch0_0.h264':
  Metadata:
    title           : Session streamed by "rRTSPServer"
    comment         : ch0_0.h264
  Duration: N/A, start: 1.243789, bitrate: N/A
  Stream #0:0: Video: h264 (Main), yuvj420p(pc, bt709, progressive), 1920x1080, 20 fps, 20 tbr, 90k tbn, 40 tbc
Stream mapping:
  Stream #0:0 -> #0:0 (h264 (native) -> h264 (libx264))
Press [q] to stop, [?] for help
[h264 @ 0x55c1ee7b5740] error while decoding MB 84 45, bytestream -15
[h264 @ 0x55c1ee7b5740] concealing 2725 DC, 2725 AC, 2725 MV errors in I frame
rtsp://192.168.32.23/ch0_0.h264: corrupt decoded frame in stream 0
[libx264 @ 0x55c1ee56b900] using cpu capabilities: MMX2 SSE2Fast SSSE3 SSE4.2 AVX FMA3 BMI2 AVX2
[libx264 @ 0x55c1ee56b900] profile High, level 4.0, 4:2:0, 8-bit
[libx264 @ 0x55c1ee56b900] 264 - core 163 r3060 5db6aa6 - H.264/MPEG-4 AVC codec - Copyleft 2003-2021 - http://www.videolan.org/x264.html - options: cabac=1 ref=3 deblock=1:0:0 analyse=0x3:0x113 me=hex subme=7 psy=1 psy_rd=1.00:0.00 mixed_ref=1 me_range=16 chroma_me=1 trellis=1 8x8dct=1 cqm=0 deadzone=21,11 fast_pskip=1 chroma_qp_offset=-2 threads=24 lookahead_threads=4 sliced_threads=0 nr=0 decimate=1 interlaced=0 bluray_compat=0 constrained_intra=0 bframes=3 b_pyramid=2 b_adapt=1 b_bias=0 direct=1 weightb=1 open_gop=0 weightp=2 keyint=250 keyint_min=20 scenecut=40 intra_refresh=0 rc_lookahead=40 rc=crf mbtree=1 crf=23.0 qcomp=0.60 qpmin=0 qpmax=69 qpstep=4 ip_ratio=1.40 aq=1:1.00
Output #0, mp4, to '/home/test.mp4':
  Metadata:
    title           : Session streamed by "rRTSPServer"
    comment         : ch0_0.h264
    encoder         : Lavf58.76.100
  Stream #0:0: Video: h264 (avc1 / 0x31637661), yuvj420p(pc, bt709, progressive), 1920x1080, q=2-31, 20 fps, 10240 tbn
    Metadata:
      encoder         : Lavc58.134.100 libx264
    Side data:
      cpb: bitrate max/min/avg: 0/0/0 buffer size: 0 vbv_delay: N/A
[rtsp @ 0x55c1ee52b680] max delay reached. need to consume packete=N/A speed=   0x
[rtsp @ 0x55c1ee52b680] RTP: missed 33 packets
[h264 @ 0x55c1ee6092c0] concealing 2678 DC, 2678 AC, 2678 MV errors in I framerop=5 speed=   0x
rtsp://192.168.32.23/ch0_0.h264: corrupt decoded frame in stream 0=N/A dup=0 drop=5 speed=   0x
[rtsp @ 0x55c1ee52b680] max delay reached. need to consume packete=N/A dup=1 drop=5 speed=   0x
[rtsp @ 0x55c1ee52b680] RTP: missed 40 packets
[h264 @ 0x55c1ee77bbc0] error while decoding MB 111 41, bytestream -22 dup=1 drop=5 speed=   0x
[h264 @ 0x55c1ee77bbc0] concealing 3178 DC, 3178 AC, 3178 MV errors in I frame
rtsp://192.168.32.23/ch0_0.h264: corrupt decoded frame in stream 0
[rtsp @ 0x55c1ee52b680] max delay reached. need to consume packette=1553.6kbits/s dup=3 drop=5 speed=0.301x
[rtsp @ 0x55c1ee52b680] RTP: missed 40 packets
[h264 @ 0x55c1ee5e6c40] error while decoding MB 21 42, bytestream -13310.9kbits/s dup=3 drop=5 speed=0.318x
[h264 @ 0x55c1ee5e6c40] concealing 3148 DC, 3148 AC, 3148 MV errors in I frame
rtsp://192.168.32.23/ch0_0.h264: corrupt decoded frame in stream 0e= 932.2kbits/s dup=3 drop=5 speed=0.406x
[rtsp @ 0x55c1ee52b680] max delay reached. need to consume packette=1215.8kbits/s dup=6 drop=5 speed=0.524x
[rtsp @ 0x55c1ee52b680] RTP: missed 40 packets
[h264 @ 0x55c1ee75ef00] left block unavailable for requested intra4x4 mode -1ts/s dup=6 drop=5 speed=0.522x
[h264 @ 0x55c1ee75ef00] error while decoding MB 0 43, bytestream 358
[h264 @ 0x55c1ee75ef00] concealing 3049 DC, 3049 AC, 3049 MV errors in I frame
rtsp://192.168.32.23/ch0_0.h264: corrupt decoded frame in stream 0e=1446.4kbits/s dup=6 drop=5 speed=0.57x
[rtsp @ 0x55c1ee52b680] max delay reached. need to consume packette=1133.6kbits/s dup=9 drop=5 speed=0.639x
[rtsp @ 0x55c1ee52b680] RTP: missed 23 packets
[h264 @ 0x55c1ee678800] left block unavailable for requested intra mode1.8kbits/s dup=9 drop=5 speed=0.642x
[h264 @ 0x55c1ee678800] error while decoding MB 0 46, bytestream 11277
[h264 @ 0x55c1ee678800] concealing 2689 DC, 2689 AC, 2689 MV errors in I frame
rtsp://192.168.32.23/ch0_0.h264: corrupt decoded frame in stream 0
[rtsp @ 0x55c1ee52b680] max delay reached. need to consume packette=1089.5kbits/s dup=11 drop=5 speed=0.71x
[rtsp @ 0x55c1ee52b680] RTP: missed 12 packets
[h264 @ 0x55c1ee725580] concealing 3095 DC, 3095 AC, 3095 MV errors in I frame
rtsp://192.168.32.23/ch0_0.h264: corrupt decoded frame in stream 0e=1286.6kbits/s dup=11 drop=5 speed=0.716x
[rtsp @ 0x55c1ee52b680] max delay reached. need to consume packette=1277.5kbits/s dup=13 drop=5 speed=0.758x
[rtsp @ 0x55c1ee52b680] RTP: missed 41 packets
[h264 @ 0x55c1ee642f40] left block unavailable for requested intra4x4 mode -1
[h264 @ 0x55c1ee642f40] error while decoding MB 0 42, bytestream 1067
[h264 @ 0x55c1ee642f40] concealing 3169 DC, 3169 AC, 3169 MV errors in I frame
rtsp://192.168.32.23/ch0_0.h264: corrupt decoded frame in stream 0e=1227.6kbits/s dup=13 drop=5 speed=0.757x
[rtsp @ 0x55c1ee52b680] max delay reached. need to consume packette=1282.1kbits/s dup=16 drop=5 speed=0.785x
[rtsp @ 0x55c1ee52b680] RTP: missed 34 packets
[h264 @ 0x55c1ee7088c0] concealing 2682 DC, 2682 AC, 2682 MV errors in I frames/s dup=16 drop=5 speed=0.793x
rtsp://192.168.32.23/ch0_0.h264: corrupt decoded frame in stream 0e=1522.1kbits/s dup=16 drop=5 speed=0.791x
[rtsp @ 0x55c1ee52b680] max delay reached. need to consume packette=1387.8kbits/s dup=19 drop=5 speed=0.813x
[rtsp @ 0x55c1ee52b680] RTP: missed 14 packets
[h264 @ 0x55c1ee7d2500] cabac decode of qscale diff failed at 13 46=1482.1kbits/s dup=19 drop=5 speed=0.807x
[h264 @ 0x55c1ee7d2500] error while decoding MB 13 46, bytestream 23335
[h264 @ 0x55c1ee7d2500] concealing 2676 DC, 2676 AC, 2676 MV errors in I frame
rtsp://192.168.32.23/ch0_0.h264: corrupt decoded frame in stream 0
[rtsp @ 0x55c1ee52b680] max delay reached. need to consume packette=1441.8kbits/s dup=21 drop=5 speed=0.836x
[rtsp @ 0x55c1ee52b680] RTP: missed 12 packets
[h264 @ 0x55c1ee6ebec0] concealing 794 DC, 794 AC, 794 MV errors in I framebits/s dup=21 drop=5 speed=0.825x
rtsp://192.168.32.23/ch0_0.h264: corrupt decoded frame in stream 0
[rtsp @ 0x55c1ee52b680] max delay reached. need to consume packette=1386.6kbits/s dup=24 drop=5 speed=0.853x
[rtsp @ 0x55c1ee52b680] RTP: missed 34 packets
[h264 @ 0x55c1ee7b5740] error while decoding MB 82 45, bytestream 203
[h264 @ 0x55c1ee7b5740] concealing 2727 DC, 2727 AC, 2727 MV errors in I frame
---------------------------------------------

roleoroleo, 2023-11-27:
Too many errors.
Please try to use tcp transport:
`ffmpeg -rtsp_transport tcp -i rtsp://IP_CAM/ch0_0.h264 test.mp4`

ibratos, 2023-12-28:
Him check this fix: https://github.com/roleoroleo/yi-hack-Allwinner/issues/419

closed 2024-05-04 by github-actions[bot]