roleoroleo/yi-hack-Allwinner #423: Status reports PTZ "no" for Dome y30qa

TheTimeWalker, 2024-01-02:
Currently, there is an issue where Dome models, specifically y30qa, where PTZ does not work for other integrations like the Home Assistant yi-hack integration.

The problem seems to be that when I check under `/cgi-bin/status.json` I see:
```json
{
  "ptz": "no"
}
```
which signals to any integration checking this status that PTZ does not work on it.

The weird thing about that is, that the Dome model does support PTZ. Not only that, but PTZ actually works fine on the web GUI. If trying to send commands to it, like with HA integration, it fails as it checks the status.json file and thinks that the camera can't use that.

I've tried finding out if there is an option connected to it but couldn't find it. It's disconnected from the cloud, RTSP stream is turned on and as a test I also turned on ONVIF. The issue wasn't there with older versions, however I can't pinpoint at what version it broke. I'm currently at the latest 0.3.7

roleoroleo, 2024-01-05:
It's a bug, you are right.
I already fixed it, but not yet commited.

closed 2024-05-15 by github-actions[bot]

TheTimeWalker, 2024-05-15:
Hey there, sorry forgot to report, but this has been fixed 👍🏻