roleoroleo/yi-hack-Allwinner #426: YI X dome (y30qa_0.3.7) - two identical cameras - one not responding

Petengy, 2024-01-07:
Hi everyone
I'll start by congratulating you roleoroleo for the splendid work.
I tested your firmware and I must say that everything went ok on the first camera, even on the second (same network but connected with repeater in mesh network) apparently... but on this one (correctly integrated like the first in HA) none of the functions has an effect (for example switch on/off camera or switch on/off cruise, etc.), only video streaming works.

The log file does not present any anomalies (it seems to me).

Some idea ?

D.

roleoroleo, 2024-01-07:
Did you configure properly mqtt?

Petengy, 2024-01-07:
..you know when you don't see the elephant in the room?

[image]

closed 2024-05-15 by github-actions[bot]