roleoroleo/yi-hack-Allwinner #427: Online FW upgrade not working?

roleoroleo, 2024-01-19:
Online FW update no longer works.
busybox needs to be updated.
In the meantime, use the manual firmware update procedure:
https://github.com/roleoroleo/yi-hack-Allwinner/wiki/Manual-firmware-upgrade

snakuzzo, 2024-01-21:
It doesn't work.
I'm on 0.3.6 and in Maintenance page there is I click on FIRMWARE UPGRADE I've got `Unable to download firmware file`

[image]

[image]

tsedgwick1, 2024-01-21:
Did you find a solution ? I have the same error message with v0.3.7 version and camera Yi 1080p Home BFUS

essys, 2024-01-21:
In the linked [manual upgrade process](https://github.com/roleoroleo/yi-hack-Allwinner/wiki/Manual-firmware-upgrade) it says:
_Rename the file replacing the version numbers with the 'x' char (y20ga_0.3.7.tgz -> y20ga_x.x.x.tgz)_

It looks as if your file still contains version designations.

snakuzzo, 2024-01-21:
Sorry, I've read wrong :D

Madaraxa, 2024-04-11:
Still no online Update possible?
I cant remember if it bricks the cam by trying or just nothing happens :)
I am on 0.3.8 and want to get the todays new 0.3.9.

Madaraxa, 2024-04-12:
I just tried it and both my cams updated from 0.3.8 to 0.3.9 without any problems.
So @roleoroleo close this as fixed?

oopiicaa, 2024-04-22:
It worked for me too.

frosty-geek, 2024-05-18:
update from 0.3.7 is still broken, will do a manual update now

closed 2024-09-16 by github-actions[bot]

koichirose, 2026-09-28:
Hi, is this broken again? Trying to update from 0.3.9 to 0.4.1.

roleoroleo, 2026-09-29:
It should work.
This bug was fixed a long time ago.

Madaraxa, 2026-09-29:
Worked for me some time ago from 0.3.9 to 0.4.0 and now to 0.4.1

koichirose, 2026-09-29:
Nevermind, my fault. I forgot I blocked internet access to the cameras 😊 Sorry!