roleoroleo/yi-hack-Allwinner #429: Incorrect time for Motion Events

Svartskog, 2024-04-03:
The correct time/date is displayed under "Status". However, as soon as a video record is triggered, the video always has an incorrect time stamp, always 2 hours earlier.

The correct time zone is set in the menu item Configurations --> Timezone
As soon as a video is recorded, e.g. at 12 o'clock, it is displayed under Motion Event at 10 o'clock and so on.

Have I forgotten a setting?

LZsolt75, 2024-04-03:
The same situation here. model: y20ga BFUSY31AYZDYA5200308. The two-hour difference came after the switch to summer time, until then it was one hour difference.

roleoroleo, 2024-04-06:
> the video always has an incorrect time stamp, always 2 hours earlier.

Where? Webgui? Yi app? Other?

> it is displayed under Motion Event at 10 o'clock and so on.

Which column is wrong? "Date & Time" or "Directory name"?
Are you using the option "Enable time OSD"?

Svartskog, 2024-04-06:
Yi App: Time ok
Webgui Status Page: Time ok
Webgui Motions Events Page: Date is ok, Time -2 hours //  Directory name: Date is ok, Time-2 hours
[image]

Enable time OSD is off

roleoroleo, 2024-04-07:
I thought I had already solved this problem.
Try this version, please:
[y25ga_0.3.8.tar.gz](https://github.com/roleoroleo/yi-hack-Allwinner/files/14896536/y25ga_0.3.8.tar.gz)

Svartskog, 2024-04-07:
a fix in the same version? since I already had the .3.8 on it
and a new Option? 
[image]
But many thanks! now everything fits :)
[image]

roleoroleo, 2024-04-07:
It's a beta version, the version number is not important.

dallas6672, 2024-06-27:
Just updated to version y20ga_0.3.9. Noticing that NTP will not sync when the cloud is Disabled. System time never updates from the Default (Fri Apr 17 20:25:35 GMT 2020)

If I enable the Cloud and then block all Ports at the firewall except NTP, time is still able to sync. But, the time is off on the Web Index page and checking "date" from command line.

Says: Wed Jun 26 06:48 EDT 2024
Actual: Wed Jun 26 11:39 EDT 2024

closed 2024-10-27 by github-actions[bot]