roleoroleo/yi-hack-Allwinner #434: Service Camera Snapshot not working Home Assistant

ManiSulClude, 2024-07-31:
Recalling service: Camera.Snapshot not work.
The image file is created but the size is always 0B

roleoroleo, 2024-07-31:
Does it work if you call it from the web gui?
I tested it now with ha and it works.
Have you tested it with a correct external dir?
Do you have the permissions to write to this folder?

ManiSulClude, 2024-07-31:
in attachment the screenshot of service recal from HA dev tool
[image]

What do you mean with "Does it work if you call it from the web gui?"

roleoroleo, 2024-08-01:
Maybe a "/" is missing before tmp?
Open this url:
`http://IP-CAM/?page=snapshot`
using the IP of your cam.

ManiSulClude, 2024-08-01:
Hi,

I verified in the web UI and the snapshot feature was disabled.
I enabled it and now it works.
I only noticed that the snapshot in hi res take so much time (about 3 seconds) and to avoid timeout in HA automation I must keep active the "Force low res snapshot feature".

We can close the issue

Best regards
Cristian

roleoroleo, 2024-08-02:
Yes, I know. The snapshot is very expensive in terms of cpu and memory.
So, it requires time.

closed 2024-08-02 by roleoroleo