roleoroleo/yi-hack-Allwinner #435: Camera buffering when motion detected and generally bad WiFi

mortenmoulder, 2024-08-08:
Running firmware 0.3.9 on a Yi 1080p (y20ga) camera with base version 8.2.0.0A_201912270941. Our WiFi is powered by a couple of Unifi U7 Pro access points. The camera has everything disabled except RTSP and the HTTP interface.

First off, the WiFi is horribly bad. If I run cnping or just regular ping on a camera literally 5 meters from an access point, the connection is so bad, that the RTSP stream is unable to keep up. Case and point:

[image]

It's like this 24/7. I have opened up my camera and replaced the factory antenna with a 2.4 GHz omni directional one, that definitely improved the stability quite a bit, but it's still bad. Going back to factory yields worse results.

Is there anything I can do, to improve the stability? I've had these for a couple of years, and I don't remember it being this bad before. It could simply be the SOC being overloaded, right?

---

The other issue I experience, is that once motion is detected, there's a high chance that the camera will simply start buffering. I can see this in the cnping chart as well, where it instantly drops the network connection. I'd say it happens more than it doesn't, so maybe 2/3 of the time motion is detected. The camera is located in a hallway.

I remember the older version having a "motion detection disable/enable" toggle, but on 0.3.9 it just says it's always on.

I really suspect the SOC is struggling to keep up, which could be related to the network connectivity as well.

---

I tried searching across all the different "yi hack" projects, but came across nothing like this. Any clue where to begin?

Indianb0y016, 2024-08-10:
If changing an antenna helped your stability, and you dont remember the cameras being this unstable before, then the next thing to check is your wifi environment. Are the 2.4ghz networks set the 20mhz channel width? What is the radio strength set to? Channels? 

The camera's SOCs really are not that great, so its probably best you leave motion detection to an nvr of sorts, rather than the camera itself. 
If you really want rock solid stability from your cameras, unfortunately youll have to get something that is wired via ethernet. Wifi is really not good for 24/7 rtsp streaming, especially on a crowded frequency (2.4Ghz).

roleoroleo, 2024-08-11:
> Is there anything I can do, to improve the stability? I've had these for a couple of years, and I don't remember it being this bad before. It could simply be the SOC being overloaded, right?

I have the same cam and it's not so bad. To check if the problem is related to the soc, try to kill all unneeded services.
For example, you could try to do a test with ftp after killing rRTSPServer.
If you kill rmm (the process that uses the most cpu) remember to kill all whatchdogs.

Another issue related to the cpu load (and maybe related to the motion detection) is the snapshot.
Try to disable snapshots or force them to low res.

Indianb0y016, 2024-08-11:
> I have the same cam and it's not so bad. To check if the problem is related to the soc, try to kill all unneeded services.
> For example, you could try to do a test with ftp after killing rRTSPServer. [...]

I realize this is unrelated, but what exactly does ./rmm process do? I got curious about the cameras system usage, and am experimenting with dual rtsp streams to evaluate the quality recordings in my nvr. As you said, rmm uses the most cpu. What does this process do exactly? Thank you!

roleoroleo, 2024-08-11:
rmm is the main process and is part of the factory firmware by Yi.
It extracts audio and video from the sensor, encode the stream, detect motions, etc...

mortenmoulder, 2024-08-11:
@roleoroleo I'm guessing it's not the CPU usage. Dang.

[image]

No clue what changed. My WiFi setup is pretty dope, if I do say so myself. Consumer++ grade. It's rock solid on any other device. It makes no sense that this one single device is so bad in terms of WiFi.

---

While writing this, I noticed something funny. I was doing a cnping while watching `top`, and I noticed that when the camera dropped the connection, A LOT of the processes were gone. Maybe it's just because of SSH packet loss, so the output is truncated. That's my best guess.

[image]

Indianb0y016, 2024-08-11:
> rmm is the main process and is part of the factory firmware by Yi.
> It extracts audio and video from the sensor, encode the stream, detect motions, etc...
>  
> 

I see. Thank you for clarifying! Thanks again for maintaining the awesome projects!

Indianb0y016, 2024-08-11:
> @roleoroleo I'm guessing it's not the CPU usage. Dang.
>
> [image]
>
> No clue what changed. My WiFi setup is pretty dope, if I do say so myself. Consumer++ grade. It's rock solid on any other device. It [...]
I dont use Ubiquiti, but rather TP-Link Omada and I also consider my network setup awesome. However, there are edge cases where devices are made without much stability in mind with the hardware/software. Some devices may advertise compatibility with what you have, and yet are just problematic to work with. I had an ESP32 device that just refused to work with one of my Omada APs, no matter what I did. It only worked after a firmware update was pushed out to the access point. So check your APs firmwares, and settings. If nothing works, and this particular camera is still working poorly, try it with a low resolution rtsp stream. I noticed the high rtsp stream caused the camera to hitch regularly with the low rtsp stream as well. I think the high stream was just too much for it. Your image shows usage of high quality stream. Try low quality and see if anything changes.

mortenmoulder, 2024-08-11:
> If nothing works, and this particular camera is still working poorly, try it with a low resolution rtsp stream. I noticed the high rtsp stream caused the camera to hitch regularly with the low rtsp [...]

I actually tried that, just to rule out a couple of things, but that didn't do anything positive, unfortunately.

I'll try tinkering with the WiFi settings once again, and hopefully get some better results, otherwise I'm buying a different camera. I've got a couple of Reolink WiFi cameras and they're rock solid. Both outside and long distance to the APs, but they're rock solid. It's so weird, but oh well.

Thanks for the help!

roleoroleo, 2024-08-15:
You should try to use a different temporary ap.
Just to check if the cam is the problem.
It could be a compatibility issue or a hardware failure of the cam.

About the top command, the truncated output could be caused by a cpu overload.
Please, try to disable all unnecessary services, primarily snapshot, if you haven't already.

closed 2024-12-14 by github-actions[bot]