roleoroleo/yi-hack-Allwinner #438: No TZ for America/Santiago (Chile)

setterlee, 2024-09-08:
Hi, I was looking the Timezone for Santiago of Chile and didn't find it...

Could you please tell me any walkarround for this or add the TZ? thanks.

roleoroleo, 2024-09-12:
The list must be updated.
Meanwhile, try CLST.

setterlee, 2024-09-14:
Hi @roleoroleo, thanks for your answer.  

the CLST set a time with plus 3 hours of the Chilean current time, I was using the Argentina's Timezone AST3 that at this moment is the same TZ here in Chile. 

Also I'm using a local NTP server to sync all my cameras time because I block the cameras internet access on my router and I noticed that all my yihacked cameras didn't sync the time while they have the internet access limited even with my local ntp server setted..

roleoroleo, 2024-09-14:
Other users noticed strange behaviors when internet access is blocked.
But this is strange: the ntp client starts in the system.sh script and it's not related to the internet access.
Are you sure that the cam can connect to the server?

setterlee, 2024-09-14:
I think that cam can reach the server, at least have ping...

[image]

roleoroleo, 2024-09-14:
Please, try this: login with a ssh session and restart the ntp server.
```
killall ntpd
ntpd -p 192.168.4.2
```

setterlee, 2024-09-14:
work well via ssh.

roleoroleo, 2024-09-14:
Ok, is cloud enabled (Disable cloud=no)?

setterlee, 2024-09-14:
is disabled...

closed 2025-01-18 by github-actions[bot]