roleoroleo/yi-hack-Allwinner #446: I bricked my 9FUS camera

tunerooster, 2024-12-21:
Are the unbrick instructions at:

https://github.com/roleoroleo/yi-hack-Allwinner/tree/master/unbrick

still valid and correct for this version (y25ga_0.3.9)  of the hack?

This version seems quite different from the older versions I am used to.

Thanks!

PS: My cam is 9FUS.

roleoroleo, 2024-12-22:
Here it is:
[home_y25ga.gz](https://github.com/user-attachments/files/18222237/home_y25ga.gz)

tunerooster, 2024-12-23:
GREAT!  But I'm confused what home_y25ga.gz is and or what to do with it.  I thought I needed something like mtdblockX.bin file(s).

What am I misunderstanding?

Many thanks for your help and my apologies for my ignorance.  :)

roleoroleo, 2024-12-23:
Yes, it's the file created by the unbrick procedure.
Derived from mtdblock3.bin
You need to unzip it in the root folder of the sdard and the bootloader will load it.

tunerooster, 2024-12-24:
I must be doing something wrong.  I am following the unbrick instructions at:
https://github.com/roleoroleo/yi-hack-Allwinner/tree/master/unbrick
I unzip the file you sent into the unbrick/y25ga directory as is.  I do not rename it to mtdblock3.bin.
I am *not* running "./build.sh factory" as I assume that builds the home_y25ga file which you sent.

Here are the details of what I did:

# fdisk -l /dev/sdd
Disk /dev/sdd: 115.84 GiB, 124383133696 bytes, 242935808 sectors
Disk model: Storage Device  
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: dos
Disk identifier: 0x00000000

Device     Boot Start       End   Sectors   Size Id Type
/dev/sdd1        2048 242935807 242933760 115.8G  b W95 FAT32

$ git clone https://github.com/roleoroleo/yi-hack-Allwinner.git
Cloning into 'yi-hack-Allwinner'...
remote: Enumerating objects: 6492, done.
remote: Counting objects: 100% (2187/2187), done.
remote: Compressing objects: 100% (836/836), done.
remote: Total 6492 (delta 1431), reused 1967 (delta 1238), pack-reused 4305 (from 1)
Receiving objects: 100% (6492/6492), 3.39 MiB | 8.73 MiB/s, done.
Resolving deltas: 100% (4252/4252), done.

$ git clone https://github.com/roleoroleo/yi-hack-Allwinner.git
Cloning into 'yi-hack-Allwinner'...
remote: Enumerating objects: 6492, done.
remote: Counting objects: 100% (2187/2187), done.
remote: Compressing objects: 100% (836/836), done.
remote: Total 6492 (delta 1431), reused 1967 (delta 1238), pack-reused 4305 (from 1)
Receiving objects: 100% (6492/6492), 3.39 MiB | 8.73 MiB/s, done.
Resolving deltas: 100% (4252/4252), done.

$ rsync -av  ../unbrick-partition-backup/home_y25ga ./yi-hack-Allwinner/unbrick/y25ga/
sending incremental file list
home_y25ga

$ md5sum ./yi-hack-Allwinner/unbrick/y25ga/home_y25ga 
cbd6a1067e9f36f63ba077853e305275  home_y25ga

$ df -h /run/media/rwk/9FUS/
Filesystem      Size  Used Avail Use% Mounted on
/dev/sdd1       116G   12M  116G   1% /run/media/rwk/9FUS

$ rsync -av ./yi-hack-Allwinner/unbrick/y25ga/ /run/media/rwk/9FUS/
sending incremental file list
./
filename
home_y25ga
mtd.txt
newhome/
newhome/app/
newhome/app/script/
newhome/app/script/wifidhcp.sh
newhome/base/
newhome/base/tools/
newhome/base/tools/extpkg.sh

$ ls -l /run/media/rwk/9FUS/
total 11904
-rw-r--r-- 1 rwk rwk        6 Dec 24 04:25 filename
-rw-r--r-- 1 rwk rwk 12058688 Dec 23 05:38 home_y25ga
-rw-r--r-- 1 rwk rwk      281 Dec 24 04:25 mtd.txt
drwxr-xr-x 4 rwk rwk    32768 Dec 24 04:25 newhome

When I insert the sd card in the camera and turn it on, the yellow led just blinks forever (over 10 minutes).  Checking the sd card after the attempt shows no files added or deleted on the card.  I.e., no log files.

I also tried with "home_y25ga" image as the *only* file in the root directory of the sd card, but got the same result, continuous yellow blinking.

Where am I going wrong?

roleoroleo, 2024-12-24:
The file I sent you is enough to start the camera in factory mode.
After the boot you should see a reboot but no files written to the card.
Are you sure your cam is a y25ga?

tunerooster, 2024-12-25:
> Are you sure your cam is a y25ga?

All I know is that it says 9FUS on the beck of the camera under the QR code.

And there seems to be multiple possibilities based on the firmware version  installed.

On the compatibility list at:

https://github.com/roleoroleo/yi-hack-Allwinner

it says that 9FUS is either y20ga or y25ga depending on the firmware version.  And on:

https://github.com/roleoroleo/yi-hack-MStar

it says 9FUS is either y203c or y25 depending on the firmware version.

I can't check the firmware version, because the camera won't boot.

So apparently it could be y25, y20 or y203c.  Do you happen to have y20 and/or y203c I could try?

Thanks again for your help!

P.S.  Originally, I did not see the other possibilities.  I just went with the first one I saw (which was y25ga) and did not look further.

roleoroleo, 2024-12-30:
y203c and y25 are MStar based, not Allwinner.
The unbrick procedure is different.
Are you able to open it and check if there is a label on the pcb?

tunerooster, 2024-12-31:
Yes!  It says:

 y20ga_mb_v1.3 2019-12-09

on the PCB.

tunerooster, 2025-01-05:
I found these two files on github, which I guessed might be what I need to unbrick my camera.

https://github.com/roleoroleo/yi-hack-Allwinner/files/12893869/mtdblock3.bin.gz
for 8.2.0.0A_201912270941
https://github.com/user-attachments/files/18232225/mtdblock3.bin.gz
for 12.2.0.2_202212191129

It appears one is using firmware  8.2.0.0A_201912270941 and the other 12.2.0.2_202212191129

I tried both using the unbrick instructions at:
github.com/roleoroleo/yi-hack-Allwinner/tree/master/unbrick
and successfully ran
```
./build.sh factory
```
but neither worked.  In both cases the yellow led went solid for a few seconds, then started to blink and never stopped blinking (even afer an hour).  It did produce a log file, which I have attached.

[log_start_reset.tar.gz](https://github.com/user-attachments/files/18310167/log_start_reset.tar.gz)

Please advise and thank you kindly!

roleoroleo, 2025-01-05:
From the log it seems that the cam is properly running but not hacked.
Try to hack it or
`./build.sh hacked`
This last command add a running telnet to the system.

tunerooster, 2025-01-06:
After running ./build factory, I installed the hack and plugged in the camera.  The led turned blu for a few seconds, then flashed blur for a few seconds, then turned yello for a few seconds, the flashed yellow for over an hour and never stopped.  I did 

I am now trying with ./build factory.

It's a unclear to me how the unbrick gets connected to wifi without an ssid and password.  That is, hou would I use ftp, until wifi is connected.

I also do not understand re: ./build factory.  I thought I had to get back to a factory default, where it speaks "waiting to connect", so I can setup wifi in the app before installing the hack.

Thanks again for your help!

tunerooster, 2025-01-06:
Woah!  Something worked!!!  Using "./build hacked", as you instructed, the camera booted and connected to wifi (I set my ssid and password in the wifi config file).

However, nothing is running except telnet, and I ***am*** able to telnet in successfully.

This is a new state to me.  I installed the Allwinner-firmware-y20ga_0.3.9 hack on the sd card after unbricking.  I was hoping to get the full hack up with this, including http.  But this is significant progress.  However, I don't know how to proceed.  Please let me know how  to get the full hack up from here.  Thanks and YAY!

tunerooster, 2025-01-07:
Somehow, the file script/system.sh is corrupt on the camera.  The first block is all zeros.  I assume this is the problem...
It is fine on the download on my pc.

Can I safely unpack home_y20gam.stage  in the root (/) on tha camera?  Or do I need to repeat the hack install from scratch?

Thanks!

tunerooster, 2025-01-07:
OK, I reinstalled the hack on a new sd card and the same thing happened.
So I unpacked the file "home_y20gam.stage" on my PC and there is no problem with system.sh there.
It appears the problem is caused by something the hack installation is doing.

So I checked some other files and found this on the camera:
```
# /home/yi-hack/bin/rRTSPServer -r -p
Error loading shared library : Invalid argument (needed by /home/yi-hack/bin/rRTSPServer)
Error loading shared library : Invalid argument (needed by /home/yi-hack/bin/rRTSPServer)
# /home/yi-hack/bin/hostname
Segmentation fault
```
I checked the md5sum results for at lease these three files which differ in the unpacked yi-hack on my PC and the resulting files on the camera after the hack is installed:
```
On the camera:
85c62e6d5194d8f6cd5e0f077c56b1de  /home/yi-hack/script/system.sh
aaf1f74dfee034a86764e7debe9be509  /home/yi-hack/bin/busybox
e094f100aca50c11c92f06984b95cf4b  /home/yi-hack/bin/rRTSPServer

On my PC:
73725e05ef578dc58dd33eb86a2fdf7f  /tmp/home/yi-hack/script/system.sh
f54d4558785156b4911a7867fe561f0c  /tmp/home/yi-hack/bin/busybox
6c9eb81e01f446d37e730fa1bea2a569  /tmp/home/yi-hack/bin/rRTSPServer
```
So something is going wrong with the hack install.

This is the md5sum of the file y20ga_0.3.9.tgz which I installed from:
```
eb3bb219d710968dfd05ef4bfef8d067  y20ga_0.3.9.tgz
```
Can you suggest what I might do to find the problem?

Thanks!

roleoroleo, 2025-01-07:
> I also do not understand re: ./build factory. I thought I had to get back to a factory default, where it speaks "waiting to connect", so I can setup wifi in the app before installing the hack.

./build factory creates the original partition when you bought the can.
./build hacked creates the same partition with 2 differences: telnet enabled and a random file that changes the final checksum.
Why this? Because the bootloader saves the hash of the partition load (to avoid to load again the same file) and if the file is not changed it doesn't load. If you try to load the factory partition but it's the same file you loaded before, ghe bootloadr will not load it. If you use hacked, the hask will change each time.

> P.S. How long should I wait while it is flashing yellow before giving up:
> 
>     1. to unbrick
> 
>     2. after installing the hack

It depends on how many files there are in the sd card.
If you clean the sd card before the update (just the update) file, a couple of minutes.

roleoroleo, 2025-01-07:
> Can you suggest what I might do to find the problem?

This problem happened to me too and I haven't solved it yet.
Unfortunately I think the cause is the flash that is now damaged (too many write cycles). But I'm not sure yet.
I checked the build folder of the project and the files are ok.
I checked the content of the update file (the hack file) and the files are ok.
I checked the update procedure and it's ok.
At the moment I don't know, but I will try again in the next days.

tunerooster, 2025-01-08:
Ok, let me know.

tunerooster, 2025-01-08:
roleo, do you happen to know if BFUS 1080p Home cameras use ARMv7 processors?

I am trying to run a 1080p Home camera off a powerbank battery.  I have found that there are power issues with cams which are **not** running ARMv7 processors.  If the flash memory  is trashed on my 9FUS, I likely need a replacement for my 9FUS.  I have found a Yi Home BFUS on ebay, but unless it is ARMv7, I don't want to risk it working with my powerbank.
For anyone who may not know,  the processor type is shown running this on the camera:
```
cat /proc/cpuinfo
```
Also, are there any issues with the allwinner-v2 hack for the BFUS 1080p?

Anyone else reading this, please feel free to reply!

Thanks!

roleoroleo, 2025-01-11:
All Mstar and Allwinner are ARMv7.
I don't remember v4/v5 butI think those are ARMv7 too.

Meanwhile I tried to investigate the problem with the cam but I didn't find the reason.
I downgraded and upgraded again with the same versione and it's working properly.
No problems during upgrde.
Probably it was a problem with the flash but not a hardware problem.
Maybe a file system problem: so if you will use the unbrick procedure you will fix the problem.
I think...

tunerooster, 2025-01-12:
I'll try it again.  Thanks for the update!

closed 2025-05-12 by github-actions[bot]