roleoroleo/yi-hack-Allwinner #447: BFUS y20ga firmware 12.2.0.5_202411131404

divadiow, 2024-12-21:
[12.2.0.5_202411131404home_y20gam.zip](https://github.com/user-attachments/files/18219938/12.2.0.5_202411131404home_y20gam.zip)

http://yihome-firmware-eu.oss-eu-central-1.aliyuncs.com/yifirmware/smarthomecam/familymonitor-y20ga/311/12.2.0.5_202411131404home_y20gam

divadiow, 2024-12-21:
[image]

divadiow, 2024-12-21:
can't actually get my cam to update to 12.2.0.5_202411131404. I've unhacked, many resets. Says it's updating successfully but eventually comes back as 12.2.0,2.

Maybe I'll solder it up and restore as per https://github.com/roleoroleo/yi-hack-Allwinner/wiki/Unbrick-the-cam#physical-connection

roleoroleo, 2024-12-22:
Let me try with my y20ga.

roleoroleo, 2024-12-22:
Yes, the update doesn't work.
I'll check it.

divadiow, 2024-12-22:
hmm OK.

just setting myself up to fix one with a different issue. Hoping to get it working and updated to latest fw too.

[image]

roleoroleo, 2024-12-22:
```
mv: write error: No space left on device
mv: can't preserve times of '/home/home_y20gam': No space left on device
mv: can't preserve ownership of '/home/home_y20gam': No space left on device
mv: can't preserve permissions of '/home/home_y20gam': No space left on device
```

This is the result. I think it's necessary to remove the hack before updating.

divadiow, 2024-12-23:
Hmm. I have unhacked. I could run unhack again I guess

divadiow, 2024-12-23:
Is there a way to flash back to 100% factory?

divadiow, 2024-12-23:
hmm. back to bootlooping after unhacking if paired

roleoroleo, 2024-12-23:
Follow the unbrick procedure.
https://github.com/roleoroleo/yi-hack-Allwinner/tree/master/unbrick

divadiow, 2024-12-27:
hmm. I used a hacked backup from a 12.0.2 cam and ran /factory build on it. the resultant file was copied to SD and the camera flashed it. still no upgrade success :/

do you have a 12.0.5 mtdblock2/3 ready to flash?

roleoroleo, 2024-12-30:
It was probably not flashed because the hash is the same.
But...
Here it is: [home_y20ga.gz](https://github.com/user-attachments/files/18277052/home_y20ga.gz)
Fully hacked 12.2.0.5 tested with my cam.
You don't need to hack it again and probably you don't need to reconfigure it (if you have a sd card).

konqueror81, 2025-04-24:
> Probabilmente non è stato flashato perché l'hash è lo stesso. Ma... Eccolo: [home_y20ga.gz](https://github.com/user-attachments/files/18277052/home_y20ga.gz) Versione 12.2.0.5 completamente hackerata, testata con la mia cam. Non è necessario modificarla di nuovo e probabilmente non è necessario riconfigurarla (se si dispone di una scheda SD).

Non riesco a fare l'aggiornamento. C'è una procedura particolare da seguire?

roleoroleo, 2025-04-25:
> Non riesco a fare l'aggiornamento. C'è una procedura particolare da seguire? Come posso utilizzare il file che hai linkato?

Copy the file to the sd card and reboot the cam.
If the model matches, the bootloader will flash the file into the home partition.

konqueror81, 2025-04-28:
> > Non riesco a fare l'aggiornamento. C'è una procedura particolare da seguire? Come posso utilizzare il file che hai linkato?
> 
> Copy the file to the sd card and reboot the cam. If the model matches, the bootloader will flash the file into the home partition.

I have to rename or unzip the file?

roleoroleo, 2025-05-05:
Unzip the file.

konqueror81, 2025-05-05:
> Unzip the file.

Nothing happens. The model is correct: BFUS y20ga

closed 2025-09-04 by github-actions[bot]