roleoroleo/yi-hack-Allwinner #448: Local time in shapshot

tunerooster, 2025-01-13:
Is there any way to have the time displayed in "local time" in the watermark of the snapshot produced by:

	http://192.168.1.86:8080/cgi-bin/snapshot.sh?res=low&watermark=yes
	
Thanks!!

maxen0815, 2025-01-19:
You can adjust the Timezone within the camera settings.
Maybe that helps?

[image]

roleoroleo, 2025-01-21:
@tunerooster 
Is you cam up to date?
There was a bug about timezone.

tunerooster, 2025-01-22:
Up-to-date factory firmware? Or yi-hack?  I searched issues for timezone but didn't see anything applicable.  It shows local time every where else, just not in snapshot.  Eg, "date" displays local time.

tunerooster, 2025-01-22:
> You can adjust the Timezone within the camera settings. Maybe that helps?
> 
> [image]

Yes, checked that... "America/Denver".  Thanks

tunerooster, 2025-01-30:
Sorry about the long delay is responding...

I am unable to find "Events time" in any of the menus.

Firmware Version	0.2.2
Base Version	9.0.26.07_202104231123
Model Suffix	h30ga
Local Time	Thu Jan 30 12:44:34 GMT 2025 (my local time is actually 5:44:34 AM)

Thanks!

roleoroleo, 2025-02-01:
Your version is very old.
You should try to update it to the new version.
I probably fixed this problem.

braxanmp, 2025-05-13:
Hi, I think I have a related problem; the time configuration is correct, and it appears correct in the snapshots, but when it creates the directory tree to upload the recordings via FTP, it creates it with two hours difference. What can it be due to? It's a y20ga, with version 0.3.9

closed 2025-09-14 by github-actions[bot]