roleoroleo/yi-hack-Allwinner #449: imggrabber fails when run at boot time - but works after 30 to 60 seconds

tunerooster, 2025-01-16:
I am trying to grab an image from the camera each time the camera is booted using imggrabber in the startup.sh script.  However, it does not work unless I wait for a certain amount of time before running imggrabber.  I would like to grab the image as soon as possible.  Is there some process, or file I can poll for its existence rather than waiting for a period of time, before running imggrabber, to assure it will work?  That is, how can I test when imggrabber will run successfully before running it?  It exits with a 0 (success) status even when it fails.  If I wait for 60 seconds, then run it, it works.  Sometimes it works after 30 seconds.  So I don't want to have to wait for a worst case time to expire.  Thanks!

roleoroleo, 2025-01-21:
imggrabber doesn't work if rmm is not properly running.
And rmm needs time to configure the codec, the sensor, etc...
You could test the header of /dev/shm/fshare_frame_buf
If the header changes the cam is ready:

root@yi-hack-allwinner:~# hexdump -C -n 32 /dev/shm/fshare_frame_buf
00000000  00 00 00 00 6f 3f 1b 00  7e 93 01 00 1a 47 09 00  |....o?..~....G..|
00000010  ab 47 09 00 26 a9 52 8a  e8 fb 0f 03 00 00 00 00  |.G..&.R.........|
00000020
root@yi-hack-allwinner:~# hexdump -C -n 32 /dev/shm/fshare_frame_buf
00000000  00 00 00 00 14 48 1a 00  7e 93 01 00 7a 5e 0d 00  |.....H..~...z^..|
00000010  66 56 0e 00 b6 b5 52 8a  a0 fc 0f 03 00 00 00 00  |fV....R.........|
00000020

tunerooster, 2025-01-30:
I found that testing for the existence of the file:
```
/tmp/iframe.idx
```
seems to work.  Once that files exists, imggrabber works.

Let me know if that makes sense.  I don't understand the significance, I just found it by trial and error.

roleoroleo, 2025-02-01:
I don't know the meaning of this file.
But if it works, it's ok.

closed 2025-06-02 by github-actions[bot]