roleoroleo/yi-hack-Allwinner #454: Could not get size of /dev/shm/fshare_frame_buf

grounddog2451976, 2025-04-14:
Hello,
when i try to use 
root@camD:~# /home/yi-hack/bin/imggrabber -r high -w > /tmp/sd/record/timelapse/snapshot.jpg
i get the error 
"Could not get size of /dev/shm/fshare_frame_buf"
The file fshare_frame_buf doesnt exist
what could be the problem?

Greetings
Alex Piehl

roleoroleo, 2025-04-19:
What  version is the cam?
And what firmware?

grounddog2451976, 2025-04-19:
Firmware Version | 0.3.9
Base Version | 8.2.0.0A_202012041528
Model Suffix | y20ga
Serial Number | BFUSY31TD21PMJ200308
Local Time | Sat Apr 19 16:09:19 CEST 2025
Uptime | 0h 0m 36s
Load Average | 0.21 0.06 0.02
Free/Total Memory | 43824/61164 KB
Free SD Space | 100%

roleoroleo, 2025-04-19:
Does the cam work with the original app?

grounddog2451976, 2025-04-19:
It does but I have no working backup.
Just two files: home_.gz and rootfs_.gz.
And I cant bring it to work again.

grounddog2451976, 2025-04-19:
Maybe the backup could bring the cam back to work. But I had no success

roleoroleo, 2025-04-20:
Usually home partition is enough to restore the cam.
Check the unbrick script to understand how to prepare the correct file:
https://github.com/roleoroleo/yi-hack-Allwinner/blob/master/unbrick/build.sh

closed 2025-08-20 by github-actions[bot]