roleoroleo/yi-hack-Allwinner #455: downgrade base firmware

dentex, 2025-05-03:
Ciao **roleoroleo**,
I have a BFUS camera with base version 9.0.19 (at the moment running a [yi-hack-Allwinner-v2](https://github.com/roleoroleo/yi-hack-Allwinner-v2)). I wanted to ask if it's possible to downgrade the base to v. 8.2.0, in order to use the version of the Allwinner hack (both hacks have BFUS support).
I'm asking because this version of the hack gives the possibility to use a fixed IP address for the camera, via the dedicated button in the UI.

[image]

or... if you know how to do this on the Allwinner-v2, that would be the same ;)

I changed the router (damn locked down Starlink gen3) and I cannot disable DHCP or reserve an IP via MAC address.
Otherwise I have to do the bypass and use the old router.

Thanks.

roleoroleo, 2025-05-04:
You can't "downgrade" from 9 to 8.
It's not the same model. BFUS it's not enough to identify the model.
But you can set manual IP address in allwinner-v2 too. The web page is exactly the same.

dentex, 2025-05-04:
> You can't "downgrade" from 9 to 8.
Copy that.

> But you can set manual IP address in allwinner-v2 too. The web page is exactly the same.
I'm fairly sure that the button, as shown in the image above, was missing for the Allwinner-v2 camera... but I'll check again later today.

Thanks.

roleoroleo, 2025-05-07:
This is my h52ga (allwinner-v2) with hack 0.3.3.

[image]

dentex, 2025-05-10:
Sorry for the delay. I updated to the latest version 0.3.3, but still no button... 

[image]

dentex, 2025-05-10:
Disregard the above... it's Firefox's fault... Chromium has it. 
Thanks for the assistance, 

[image]